lintweaker/xmos-native-dsd #1: DSD native for JLsounds boards

Lubaka22, 2015-05-24:
Hi Lintweaker,

JLsounds converter boards supports DSD native up to DSD256. 
Could you please include it to the list of supported converters if it's possible?
The USB ID of JLsounds is 20B1:2023

Kind regards
Lyuben

lintweaker, 2015-05-24:
Hi Lyuben,

I am happy to add the JLsounds board. Can you test?  What do you need for testing?

Lubaka22, 2015-05-25:
Hi Lintweaker,

Thank you for your prompt answer.
I tested JLsounds board on Win up to DSD256 native, without any issues. I also tested with Daphile up to DSD128 native.
I'll need only conformation that JLsounds board is supported, then I'll follow your tutorial and inform you for the result.

Kind regards
Lyuben

lintweaker, 2015-05-26:
Hi Lyuben,
If you can provide me with the output of:
cat /proc/asound/<your device>/stream0

I can whip up a patch for you to test (together with my version of MPD or my python script)

Lubaka22, 2015-05-28:
Hi Lintweaker,

Here it is:

XMOS JLsounds Hi-Rez Audio 2.0 at usb-0000:00:13.5-6, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

Thank you.

Kind regards,
Lyuben

lintweaker, 2015-05-29:
Hi Lyuben,

I've added a patch for you to test. When you get it working, can you sent the output of 
'cat /proc/asound/<your card>/pcmp0/sub0/hw_params'  ?

closed 2015-06-14 by lintweaker