lintweaker/xmos-native-dsd #2: Debian/Ubuntu kernel patches compatibility?

valqk, 2015-06-10:
Hello,
I'm going to try out to listen to DSD on my ubuntu, but before I start patching my kernel I want to ask if these patches are compatible with the Debian/Uubuntu sources?
I have Debian and Ubuntu machines for the test.
I'll use JLSounds XMOS board.
The question may be stupid, but as far as I see there are only RPMs in the repo, so I wanted to ask before I start.
Any ideas? Someone did this?

storrgie, 2015-06-11:
@valqk:
 -  what kernel are you on? (uname -r)
 - are you familiar with [mpd](http://www.musicpd.org/)?

I personally would suggest using mpd for playback, however it might be a bit of a hurdle if its your first time using it. Read about what it is so you understand what you're setting up, then I think [this link](http://www.musicpd.org/doc/user/dsd.html) is what you're looking for when setting up the config.

It appears that the support for native dsd from this repository is in the mainline for recent kernels. You should have a good time as long as your device is supported and your kernel is modern enough to have the xmos-native-dsd in it.

valqk, 2015-06-11:
I'm using mpd for playback.
At this very moment I'm using Debian's 3.16.0-4-amd64
I'm just asking If I can apply patches successfully to debian/ubuntu kernels or they won't be compatible with the diffs. Also If I can't are they applicable on the fly without changes to vanilla kenrels?

lintweaker, 2015-06-14:
Hello,
I think kernel 3.16 itself is too old, I am not sure if Debian uses a vanilla kernel or include patches from newer kernels. Besides kernel patches you also need a recent (1.0.29) ALSA or a patched 1.0.28.

closed 2015-07-28 by lintweaker