lintweaker/xmos-native-dsd #3: Need add support for Gustard X20

xxxbugxxxx, 2015-08-12:
XMOS U8 chip

USB ID  20b1:000a XMOS Ltd 

Currently not working . Thank you very much

lintweaker, 2015-08-13:
Hello,

Each device has to be added to the linux kernel individually. There is no way to figure out which device supports native DSD and how.

Can you post the output from:

cat /proc/asound/Audio/stream0

This gives me the info needed to create a patch. Can you test this?

xxxbugxxxx, 2015-08-14:
sure, here it is. 

Thank you very much

root@volumio:~# cat /proc/asound/card1/stream0
XMOS xCORE USB Audio 2.0 at usb-3f980000.usb-1.5, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S16_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

xxxbugxxxx, 2015-08-14:
closed by mistake..

xxxbugxxxx, 2015-08-17:
Any update ? Thank you veyr much

lintweaker, 2015-08-21:
I have added the patch, can you test?

xxxbugxxxx, 2015-08-21:
Thank you very much!! let me test it.

xxxbugxxxx, 2015-08-22:
tested, compiled in a rpi2 4.1.6

xxxbugxxxx, 2015-08-22:
Compile on Linux volumio 4.1.6-v7+ #810 SMP PREEMPT .

And compiled mpd with your patch of 0.19.9 and 0.19.10. Both have the same problem 

When play DSD file, DAC recognize DSD format , but DSD just stopped. 

checking MPD log shows the following item

"mpd: pcm_misc.c:620: snd_pcm_format_set_silence: Assertion `0' failed."

lintweaker, 2015-08-22:
So the kernel patch works for your DAC. The MPD log you showed does not seem to be specific to DSD playback. Please try my mpd 0.18 version to be sure. I am going to close this issue as the kernel part works and this repo is about adding native DSD support to the kernel not MPD.

closed 2015-08-22 by lintweaker

xxxbugxxxx, 2015-08-22:
Got it, Thank you very much!:)

lintweaker, 2015-08-22:
Sorry too hear. Maybe your DAC is not happy with this little kernel patch alone. Have you tried if MPD works with DoP output? If that works at least it is not a Pi USB issue.

xxxbugxxxx, 2015-08-23:
Dop works perfectly up to dop128. Also dac displays correct signal even shows dsd256.

So I guess hate path works, just some other things it does not like. Btw except for your MPD implementation any other pkg can make use of the Alsa kernel for native dsd?

Is there any value to test it in a different Linux distribution say arch Linux?

Thank you very much again;)

Also x86 platform?

lintweaker, 2015-08-23:
If you can test on a x86 (x86_64) platform...besides patching the kernel you'll need ALSA 1.0.29 or a patched version of 1.0.27/1.0.28 for 32-bit DSD sample format support.
Fedora 22 and up has ALSA 1.0.29 included.

xxxbugxxxx, 2015-08-23:
Thank you very much, will try that then.

xxxbugxxxx, 2015-08-24:
BTW, does your MPD package require libmpdclient version 2.10? seems the libmpdclient introduce did in 2.10

xxxbugxxxx, 2016-04-11:
still has the same problem after testing several different distributions on rpi2

xxxbugxxxx, 2016-04-13:
update using mpd 0.2. issue gone.