lintweaker/xmos-native-dsd #4: Support for aune X1S USB DAC and C-Media Electronics Inc. Hegel USB Audio B

c--w, 2015-11-09:
Did I understood correctly that each device has to be explicitly added to your patch?

I have:
Linux 4.1.10-rt11-1-rt #1 SMP PREEMPT RT  (compiled few days ago) (Arch Linux)
I have your latest 0.19.10 patch for MPD (btw it doesn't compile because I have automake 1.15 and it expects 1.14 so I had to tweak this)
My outputs are: 

C-Media Electronics Inc. Hegel USB Audio B at usb-0000:00:1d.0-1.7, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 2
    Packet Size = 156
    Momentary freq = 192000 Hz (0x18.0000)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S24_3LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us

and

aune X1S USB DAC at usb-0000:00:1d.0-1.7, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 392
    Momentary freq = 352804 Hz (0x2c.19b8)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

In mpd log I get: 
Nov 09 09:21 : alsa_output: opened hw:0,0 type=HW
Nov 09 09:21 : alsa_output: format=S32_LE (Signed 32 bit Little Endian)
Nov 09 09:21 : alsa_output: buffer: size=90..131072 time=255..371520
Nov 09 09:21 : alsa_output: period: size=45..65536 time=127..185760
Nov 09 09:21 : alsa_output: default period_time = buffer_time/4 = 371519/4 = 92879
Nov 09 09:21 : alsa_output: buffer_size=131072 period_size=32768
Nov 09 09:21 : output: opened plugin=alsa name="Card0" audio_format=352800:32:2
Nov 09 09:21 : output: converting from 352800:dsd:2

for aune and

Nov 09 09:16 : alsa_output: opened hw:0,0 type=HW
Nov 09 09:16 : alsa_output: format=S24_3LE (Signed 24 bit Little Endian in 3bytes)
Nov 09 09:16 : alsa_output: buffer: size=48..174762 time=250..910219
Nov 09 09:16 : alsa_output: period: size=24..87381 time=125..455110
Nov 09 09:16 : alsa_output: default period_time = buffer_time/4 = 500000/4 = 125000
Nov 09 09:16 : alsa_output: buffer_size=96000 period_size=24000
Nov 09 09:16 : libsamplerate: setting samplerate conversion ratio to 0.54
Nov 09 09:16 : output: opened plugin=alsa name="Card0" audio_format=192000:24:2
Nov 09 09:16 : output: converting from 352800:dsd:2

for Hegel

Playing the same file.

What should I do? Where to look next?
Is there a way to get alsa drivers detailed log to see why it insists on conversion?

Thanks for your work on the patch!

Regards
Željko Vranić
(zeljkovranic at gmail)

hifi25nl, 2015-11-09:
For mpd-dsd I have made a working package for archlinux here
https://aur.archlinux.org/packages/mpd-dsd/

It will install automake-1.14 from AUR

hifi25nl, 2015-11-09:
I would add my working configuration for mpd (output device section)
1) For native non DoP

audio_output {
  type "alsa"
  name "DSD native"
  device "hw:1,0"
  dsd_usb "no"
  # Enable native DSD playback
  dsd_native "yes"
  # Select 32-bit DSD_U32_LE output format
  dsd_native_type "2"
  auto_resample 	"no"
  mixer_type      "disabled"
}

2) For DoP

audio_output {
	type		"alsa"
	name		"DSD DoP"
	device		"hw:1,0"	# optional
	auto_resample 	"no"
	auto_format     "no"
	mixer_type      "disabled"
	dsd_usb		"yes"
}

c--w, 2015-11-09:
I used exactly that link to patch MPD and it didn't work. Probably because you can't force pacman to downgrade from automake 1.15. Nevermind that - it worked without errors when I untar-ed your package, replaced 3 references of 1.14 with 1.15 in makefile and packed it again.

As for mpd.conf, I changed it so many times and used all combinations possible and I found that dsd_native and dsd_native_type properties on so many links that I forgot to include it being so obvious to me now. Anyway here is my conf:
audio_output {
  type  "alsa"
  name  "Card0"
  device  "hw:0,0"
  mixer_type  "none"
  dop  "no"
  dsd_native "yes"
  dsd_native_type  "2"
  use_mmap  "yes"
  auto_resample  "no"
  auto_channels  "no"
  auto_format  "no"
}

What else?

lintweaker, 2015-11-09:
@c--w ,
Yes, for native DSD each DAC has to specifically added as there is no way to determine if and how a USB DAC supports native DSD.
For your 2 DACs it looks good. If you send me the output of lsusb I can make patches for both DACS.
Hopefully CMEDIA uses the same sample format as XMOS for native DSD.

c--w, 2015-11-09:
Ok then.
lsusb
Bus 002 Device 008: ID 20b1:3023 XMOS Ltd
Bus 002 Device 004: ID 8087:07da Intel Corp.
Bus 002 Device 003: ID 062a:0201 Creative Labs Defender Office Keyboard (K7310) S Zodiak KM-9010
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Bus 002 Device 009: ID 0d8c:0316 C-Media Electronics, Inc.
Bus 002 Device 004: ID 8087:07da Intel Corp.
Bus 002 Device 003: ID 062a:0201 Creative Labs Defender Office Keyboard (K7310) S Zodiak KM-9010
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

aune is obviously the XMOS type
Hegel is CMEDIA
We also have device of the third type - MCS
Bus 002 Device 013: ID 16d0:071a MCS
Bus 002 Device 004: ID 8087:07da Intel Corp.
Bus 002 Device 012: ID 062a:0201 Creative Labs Defender Office Keyboard (K7310) S Zodiak KM-9010
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Which gives mpd log identical to aune (S32_LE)

Do I understood correctly now that you can provide me a kernel patch? I'm not sure how would I integrate it with rt patch that I already use: https://aur.archlinux.org/packages/linux-rt/ ?
(My linux/makefile skills are rudimentary - if non-existent)
For a moment this morning I intended to reboot to fallback image thinking that rt patch could have maybe screwed up alsa DSD part but it seems far fetched...

lintweaker, 2015-11-09:
Thanks,

So 'Aune XS1' is 0d8c:0316
and the Hegel is 0d8c:0316
Can you give me the full names of the Hegel DAC?  So I can create a nice patch with the official names in them, especially for the Hegel, they have more DACs.

What is the MCS device you mention?

Wrt DoP output, you need to use 'dsd_usb' set to 'yes' if you use my version of MPD.

c--w, 2015-11-09:
Hegel HD12 DSD
Aune X1S 32BIT/384 DSD DAC

Third box is our own custom made but it uses amanero usb interface but let's forget it for now...

Thanks, I didn't know your patch uses 'dsd_usb' still. I will try now.

c--w, 2015-11-09:
I tried dsd_usb "yes"
but in mpd output I get:
Nov 09 11:39 : config: option 'dsd_usb' on line 14 was not recognized
Nov 09 11:39 : config: option 'dsd_native_type' on line 16 was not recognized

Is it possible that I actually don't have patched mpd? I checked /usr/bin and I have newly generated mpd executable. How can I check this?

lintweaker, 2015-11-09:
You probably do not have my version of MPD. My version should show something like:
[code][kramer@orwell alsa-sound-git]$ mpd -V
Music Player Daemon 0.18.23-dsd-rt[/code]

If you are using the upstream version of MPD you can use DoP just fine with 'dop'  set to 'yes' 
http://www.musicpd.org/doc/user/output_plugins.html

Wrt to your custom DAC using the Amanero USB solution, last time I checked it needed specific driver support so native DSD wont work on Linux. If they changed their firmware then maybe.

c--w, 2015-11-09:
My mpd -V output is:
Music Player Daemon 0.19.10-dsd
I used dsd-usb package for arch-linux: https://aur.archlinux.org/packages/mpd-dsd/
provided by hifi25nl who chipped in at the beginning of this thread. 
So, patch should be in effect, just not recognizing the aforementioned option? I also used dop "yes" at the same time.

More details for our devices if you need them:
Hegel HD12 (native DSD) uses Cmedia CM6632A USB 2.0 high-speed audio processor.
(http://www.cmedia.com.tw/ProductsDetail/C1Serno-1/C2Serno-8/C3Serno-12/PSerno-48.html)
 
AUNE uses XMOS USB receiver chip - xCORE-AUDIO HiRes (second generation XMOS with native DSD support).
(http://www.xmos.com/products/silicon/xcore-audio/hires)
 
Amanareo "OEM Combo384 Module" enables DSD64 i128 preko USB-a na Linux (with UAC2 compliant kernel).
 (http://amanero.com/)

c--w, 2015-11-09:
I found this:
http://www.gossamer-threads.com/lists/linux/kernel/2255270

So, this patch is just to indicate in quirks.c that found device supports DSD native? Can I add it myself with values you mentioned? (just for myself - you can publish these patches as you intended)

lintweaker, 2015-11-09:
I have created a patch for the Aune and sent it to alsa-devel.
http://mailman.alsa-project.org/pipermail/alsa-devel/2015-November/100279.html
Next up the patch for the Hegel

lintweaker, 2015-11-09:
Patch for Aune X1S is accepted. I have upload a experimental patch for the Hegel to this repo. Should be applied on top of the X1S patch.
Please test a this is the first patch for a C-Media based device.

c--w, 2015-11-10:
Hi again. I can confirm that Aune patch works fine but Hegel does not want to play DSD files. It's display flashes DSD and 128 (Not sure what that means but it surely means it's complaining about what it receives). Maybe for this device we should put SNDRV_PCM_FMTBIT_DSD_U16_LE?
It's indicative that Hegel always play at maximum S24_3LE for PCM play. See also the contents of stream0:

C-Media Electronics Inc. Hegel USB Audio B at usb-0000:00:1d.0-1.7, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S24_3LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000
    Data packet interval: 125 us

Another curious thing is that In mpd.log for all sample files that I have it says:
output: opened plugin=alsa name="Card0" audio_format=352800:dsd:2
or
output: opened plugin=alsa name="Card0" audio_format=705600:dsd:2
and from above it can be seen that Hegel supports up to 192000 sample rate. I don't know if and how I can force alsa or mpd to play at 176400:dsd:2?!

I will go now to try to compile kernel with: 
case USB_ID(0x0d8c, 0x0316): /* Hegel HD12 DSD */ 
if (fp->altsetting == 3)
            return SNDRV_PCM_FMTBIT_DSD_U16_LE;
        break; 
and see what happens then... (unless it is a stupid thing to try and I grossly misunderstood something - feel free to tell me that right away, compiling lasts for couple of hours...)

lintweaker, 2015-11-10:
Hi,

MPD (neither my version or the official) does not support using the 16-bit DSD sample formats. Only 8-bit (official and mine) and 32-bit formats (my version).
If you are building a kernel with my patches you can change the return code 'return SNDRV_PCM_FMTBIT_DSD_U32_BE' to 'return <sample fmt1> | <sample fm2>' etc so it will allow multiple sample formats at once.

It is possible the Hegel needs a special command to switch between PCM and DSD playback.

c--w, 2015-11-12:
The manual states that Hegel clicks audibly while changing to DSD and that happens when DSD_U32_BE is the format sent to it. 
Meanwhile, I tried with DSD_U16_LE and DSD_U8 but neither works. Hegel does not recognize what it receives and I get only noise.
So, it seems that Hegel HD12 DSD cannot play native DSD in any way offered by MPD+patch.
Maybe it would be best if you revoke your HEGEL patch.
Aune patch works ok.
Thanks!

lintweaker, 2015-11-12:
Too bad. If I was able to check out what the Windows driver is doing to switch between PCM and DSD it would be possible to do the same on Linux.
I did not sent in the patch for the Hegel so no problem there.

closed 2015-11-12 by lintweaker