lintweaker/xmos-native-dsd #5: Oppo HA-1 support

afalout, 2015-11-21:
http://oppodigital.com/headphone-amplifier-ha-1/

Is Xmos based: http://www.xmos.com/blog/xmos/post/customer-focus-oppo-digital

What do I need to provide to get it supported? (DoP works OK)

Thanks!

lintweaker, 2015-11-21:
I need the output of:
lsusb
cat /proc/asound/<your card>/stream0

That should provide enough info.

afalout, 2015-11-25:
Thanks for your help!

lsusb
Bus 001 Device 003: ID 0781:5583 SanDisk Corp. 
Bus 001 Device 004: ID 22d9:0416  <<< this is Oppo
Bus 001 Device 002: ID 05e3:0716 Genesys Logic, Inc. USB 2.0 Multislot Card Reader/Writer
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

cat /proc/asound/card1/stream0
OPPO OPPO HA-1 USB AUDIO 2.0 DAC at usb-0000:00:1d.7-7, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

BTW, I wanted to try to add it myself, but could not find "quirks" file to add it to. In Fedora 23... ?

Thanks!

lintweaker, 2015-11-25:
I have added the patch for the Oppo HA-1. Let me know if it works.

afalout, 2015-11-28:
Thanks Jurgen, unfortunately the kernel patches fail to apply on current Fedora 23 kernel tree:

INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-add-dsd-u32-le-v5.patch' ...
checking file include/sound/pcm.h
Hunk #1 FAILED at 183.
1 out of 1 hunk FAILED
checking file include/uapi/sound/asound.h
Hunk #1 FAILED at 219.
1 out of 1 hunk FAILED
checking file sound/core/pcm.c
Hunk #1 FAILED at 215.
1 out of 1 hunk FAILED
checking file sound/core/pcm_misc.c
Hunk #1 succeeded at 160 with fuzz 2 (offset 12 lines).
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/0001-add-native-DSD-support-for-XMOS-based-DACs.patch' ...
checking file sound/usb/quirks.c
Hunk #1 succeeded at 1284 with fuzz 1 (offset 110 lines).
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-usb-marantz-ctl-msg-quirk-v2.patch' ...
checking file sound/usb/quirks.c
Hunk #1 FAILED at 1146.
1 out of 1 hunk FAILED
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-add-dsd-be-formats.patch' ...
checking file include/sound/pcm.h
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 187 (offset 3 lines).
Hunk #2 FAILED at 206.
1 out of 2 hunks FAILED
checking file include/uapi/sound/asound.h
Hunk #1 FAILED at 220.
1 out of 1 hunk FAILED
checking file sound/core/pcm.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 215 with fuzz 1 (offset -1 lines).
checking file sound/core/pcm_misc.c
Hunk #1 succeeded at 160 with fuzz 1 (offset 8 lines).
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-usb-switch-xmos-dsd-quirk-to-be.patch' ...
checking file sound/usb/quirks.c
Hunk #1 FAILED at 1193.
1 out of 1 hunk FAILED
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-usb-add-marantz-dsd-quirk.patch' ...
checking file sound/usb/quirks.c
Hunk #1 succeeded at 1284 with fuzz 1 (offset 80 lines).
INFO - ====== Patching with '/root/rpmbuild/SOURCES/kernel/alsa-usb-marantz-select-mode-quirk-v5.patch' ...
checking file sound/usb/pcm.c
Reversed (or previously applied) patch detected!  Assume -R? [n] y
checking file sound/usb/quirks.c
Hunk #1 succeeded at 1171 with fuzz 1 (offset 69 lines).
checking file sound/usb/quirks.h
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 33 (offset 2 lines).

Is there anything I can do to fix it?

Thanks!

lintweaker, 2015-11-28:
For F23 you do not have to apply all the patches. 
All the other patches are already included in the mainline kernel. Only 
0001-usb-Add-native-DSD-support-for-Aune-X1S.patch
0001-usb-Add-native-DSD-support-for-OPPO-HA-1.patch
should be applied

afalout, 2015-12-03:
First the god news: I managed to get Oppo recognised as Native DSD device:

 playdsd.py -l
Available sound cards:
'DAC'           : USB device with native DSD support
'Intel'         : Not a (UAC2) USB sound card

Bad news: it took me several days of debugging and recompiling several version of kernel to get this working, because the device string in the patch is wrong:

     case USB_ID(0x22d8, 0x0416): /* OPPO HA-1*/ <<< WRONG

Correct, as per my post above is:

      Bus 001 Device 004: ID 22d9:0416 <<< this is Oppo

Now I have a few questions:

1) Does format SPECIAL indicate DSD native capability? Or is there another way to recognise when USB dev supports native DSD and/or DoP? (without using playdsd.py -l) Or is that DSD_* format for native and SPECIAL for DoP?

2) Is there a way for alsa kernel driver to read quirks or equivalent file at run-time so user does not need to recompile and replace kernel for this to work? That would make adding new devices many times simpler...

4) I use forked MPD based on version 0.20 with sacdiso and dvdaiso support. It is not clear 
 	from your readme weather the native DSD patches from your tree where committed to 
 	official MPD 0.18 or do I have to use your forked tree? If later, when can we hope
 	to have your patches committed upstream? (For me, it's a "catch 22" - I need SACDISO
 	support to have DSD music to play, but cant play it as DSD (in native mode) without
 	your patches...

Many thanks for your work, and your help!

Andrej

lintweaker, 2015-12-03:
Sorry about the wrong USB ID. I did update the patch last weekend.
Unfortunately the is no  sure way to know if a USB DAC support DoP, native DSD or both. The SPECIAL sample format has no relation to DoP as DoP just uses regular PCM to transport DSD.

I would hope the forked MPD does include my native DSD patches. Upstreaming patches to MPD costs me a lot of (negative) energy. Not on my shortlist of TODO's currently.

Can you confirm native DSD playback actually works for the Oppo?

afalout, 2015-12-04:
Tried patching current MPD git (so I can confirm native DSD working). Fails only on :

checking file src/output/plugins/AlsaOutputPlugin.cxx
Hunk #2 succeeded at 69 (offset 1 line).
Hunk #3 FAILED at 167.
Hunk #4 succeeded at 298 (offset 17 lines).
Hunk #5 succeeded at 677 with fuzz 1 (offset 17 lines).
Hunk #6 FAILED at 723.
Hunk #7 FAILED at 734.
3 out of 7 hunks FAILED

(The rest fail because year 2015 in Copyright line)

Some methods changed names, so did not want to mess things up by trying to fix this myself.
Any chance you can take a look?

I'm using the SACDISO tree: http://git.musicpd.org/cgit/manisiutkin/mpd.git/

Which fork did you refer to?

BTW, I know exactly what you mean by negative energy ... :(

lintweaker, 2015-12-06:
Can you try a free DSD DSF file download with playdsd.py to see if native DSD actually works with the kernel patch?

E.g.:
http://download.oppodigital.com/hra/davidelias/08%20-%20David%20Elias%20-%20Crossing%20-%20Morning%20Light%20Western%20Town%20%28DSD64%202.0%29.dsf

http://www.2l.no/hires/

closed 2016-01-11 by lintweaker