lintweaker/xmos-native-dsd #6: Oppo 105D

tommy99, 2015-12-13:
Oppo 105D seems to have the same XMos interface as HA-1, but the USB ID (22d9:0451) is different.

Could you add this USB-ID too xmos-native-dsd too?

Thank you much and Liebe Grüße

tommy99, 2015-12-13:
Here is the output of cat /proc/asound/card1/stream0

OPPO OPPO USB AUDIO 2.0 DAC at usb-bcm2708_usb-1.5, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

lintweaker, 2015-12-13:
Are you sure it support native DSD? The output of stream0 only shows the PCM interface.

closed 2016-01-11 by lintweaker