lintweaker/xmos-native-dsd #8: Ustars Audio XMOS AK4495 dsd DAC

norrest, 2016-02-03:
Hi!

```
lsusb
Bus 002 Device 002: ID 20b1:2012 XMOS Ltd
```
`Ustars Audio at usb-1c1c000.usb-1, high speed : USB Audio`

If do not edit quirks.c  I hear noise when switching modes DSD Native mode 0/2 

If i add

```
	/* XMOS based USB DACs */
	switch (chip->usb_id) {
	case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
	case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
	case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */
	case USB_ID(0x20b1, 0x2012): /* ULTRAS DAC */
		if (fp->altsetting == 2)
			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
		break;

	case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
	case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
	case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
	case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
	case USB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */
			if (fp->altsetting == 3)
			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
		break;
	default:
		break;
	}
```
section 
`case USB_ID(0x20b1, 0x2012): /* ULTRAS DAC */`

when dsd native mode = 0 

```
Ustars Audio at usb-1c1c000.usb-1, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 392
    Momentary freq = 384001 Hz (0x30.0008)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
002/002
20b1:2012
USB Mixer: usb_id=0x20b12012, ctrlif=0, ctlerr=0
Card: Ustars Audio at usb-1c1c000.usb-1, high speed
  Unit: 10
    Control: name="Ustars Audio Output Playback Volume", index=1
    Info: id=10, control=2, cmask=0x0, channels=1, type="S16"
    Volume: min=-32512, max=0, dBmin=-12700, dBmax=0
  Unit: 10
    Control: name="Ustars Audio Output Playback Volume", index=0
    Info: id=10, control=2, cmask=0x3, channels=2, type="S16"
    Volume: min=-32512, max=0, dBmin=-12700, dBmax=0
  Unit: 10
    Control: name="Ustars Audio Output Playback Switch", index=1
    Info: id=10, control=1, cmask=0x0, channels=1, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
  Unit: 10
    Control: name="Ustars Audio Output Playback Switch", index=0
    Info: id=10, control=1, cmask=0x3, channels=2, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
card: 0
device: 0
subdevice: 0
stream: PLAYBACK
id: USB Audio
name: USB Audio
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 0
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 384000 (384000/1)
period_size: 32768
buffer_size: 131072
card: 0
device: 0
subdevice: 0
stream: PLAYBACK
id: USB Audio
name: USB Audio
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 0
state: RUNNING
owner_pid   : 4927
trigger_time: 6609.441464067
tstamp      : 0.000000000
delay       : 107696
avail       : 23808
avail_max   : 129024
-----
hw_ptr      : 1400079
appl_ptr    : 1507343
tstamp_mode: NONE
period_step: 1
avail_min: 32768
start_threshold: 98304
stop_threshold: 131072
silence_threshold: 0
silence_size: 0
boundary: 1073741824

```

**Any ideas?
**

norrest, 2016-02-14:
```
aplay
Usage: aplay [OPTION]... [FILE]...

-h, --help              help
    --version           print current version
-l, --list-devices      list all soundcards and digital audio devices
-L, --list-pcms         list device names
-D, --device=NAME       select PCM by name
-q, --quiet             quiet mode
-t, --file-type TYPE    file type (voc, wav, raw or au)
-c, --channels=#        channels
-f, --format=FORMAT     sample format (case insensitive)
-r, --rate=#            sample rate
-d, --duration=#        interrupt after # seconds
-M, --mmap              mmap stream
-N, --nonblock          nonblocking mode
-F, --period-time=#     distance between interrupts is # microseconds
-B, --buffer-time=#     buffer duration is # microseconds
    --period-size=#     distance between interrupts is # frames
    --buffer-size=#     buffer duration is # frames
-A, --avail-min=#       min available space for wakeup is # microseconds
-R, --start-delay=#     delay for automatic PCM start is # microseconds
                        (relative to buffer size if <= 0)
-T, --stop-delay=#      delay for automatic PCM stop is # microseconds from xrun
-v, --verbose           show PCM structure and setup (accumulative)
-V, --vumeter=TYPE      enable VU meter (TYPE: mono or stereo)
-I, --separate-channels one file for each channel
-i, --interactive       allow interactive operation from stdin
-m, --chmap=ch1,ch2,..  Give the channel map to override or follow
    --disable-resample  disable automatic rate resample
    --disable-channels  disable automatic channel conversions
    --disable-format    disable automatic format conversions
    --disable-softvol   disable software volume control (softvol)
    --test-position     test ring buffer position
    --test-coef=#       test coefficient for ring buffer position (default 8)
                        expression for validation is: coef * (buffer_size / 2)
    --test-nowait       do not wait for ring buffer - eats whole CPU
    --max-file-time=#   start another output file when the old file has recorded
                        for this many seconds
    --process-id-file   write the process ID here
    --use-strftime      apply the strftime facility to the output file name
    --dump-hw-params    dump hw_params of the device
    --fatal-errors      treat all errors as fatal
Recognized sample formats are: S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE IEC958_SUBFRAME_LE IEC958_SUBFRAME_BE MU_LAW A_LAW IMA_ADPCM MPEG GSM SPECIAL S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE G723_24 G723_24_1B G723_40 G723_40_1B DSD_U8 DSD_U16_LE
Some of these may not be available on selected hardware
The available format shortcuts are:
-f cd (16 bit little endian, 44100, stereo)
-f cdr (16 bit big endian, 44100, stereo)
-f dat (16 bit little endian, 48000, stereo)
```

I didnt find DSD_U32_BE 

```
aplay --version
aplay: version 1.0.29 by Jaroslav Kysela <perex@perex.cz>
```

norrest, 2016-02-14:
```
grep VERSION_STR /usr/include/alsa/version.h
#define SND_LIB_VERSION_STR     "1.0.29"
```

```
 cat /proc/asound/version
Advanced Linux Sound Architecture Driver Version k4.4.1-rt5.
```

norrest, 2016-02-15:
```
/* XMOS based USB DACs */
	switch (chip->usb_id) {
	case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
	case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
	case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */
	case USB_ID(0x22d8, 0x0416): /* OPPO HA-1*/
	case USB_ID(0x20b1, 0x2012): /* ULTRAS DAC*/
			if (fp->altsetting == 2)
			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
		break;

	case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
	case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
	case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
	case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
			if (fp->altsetting == 3)
			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
		break;
	default:
		break;
	}

```
please add case USB_ID(0x20b1, 0x2012): /* ULTRAS DAC*/

DSD NATIVE WORKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

access: MMAP_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100

lintweaker, 2016-02-16:
Good to see you got it working.
Do you have a link to the specs of the DAC and the full name (Vendor and type)?

norrest, 2016-02-18:
This is China "Ustars 384 bit DSD Audio DAC" AK4495S chip.
http://vamp.obu4enie.kiev.ua/wp-content/uploads/2015/09/04.png
PS. But it has click/pop at rewind music. Nevermind dsd native or dop. (

norrest, 2016-04-21:
**Hi! Can u add case USB_ID(0x20b1, 0x2012): /* ULTRAS DAC*/ in official kernel source? **

norrest, 2016-04-21:
Hi)

closed 2016-07-20 by norrest