lintweaker/xmos-native-dsd #9: Aune T1se

rhoeden, 2016-02-20:
I’ve got Aune T1se (this is the third iteration of T1 that was released recently):

[http://en.auneaudio.com/index.php?s=Home/Article/detail/id/124.html]

and recent mpd-dsd from https://aur.archlinux.org/packages/mpd-dsd/ :

```
$ mpd -V

Music Player Daemon 0.19.12-dsd

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright (C) 2008-2014 Max Kellermann <max@duempel.org>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local smbclient nfs

Neighbor plugins:
 smbclient upnp

Decoders plugins:
 [mad] mp3 mp2
 [vorbis] ogg oga
 [oggflac] ogg oga
 [flac] flac
 [opus] opus ogg oga
 [sndfile] wav aiff aif au snd paf iff svx sf voc w64 pvf xi htk caf sd2
 [audiofile] wav au aiff aif
 [dsdiff] dff
 [dsf] dsf
 [faad] aac
 [wavpack] wv
 [modplug] 669 amf ams dbm dfm dsm far it med mdl mod mtm mt2 okt s3m stm ult umx xm
 [ffmpeg] 16sv 3g2 3gp 4xm 8svx aa3 aac ac3 afc aif aifc aiff al alaw amr anim apc ape asf atrac au aud avi avm2 avs bap bfi c93 cak cin cmv cpk daud dct divx dts dv dvd dxa eac3 film flac flc fli fll flx flv g726 gsm gxf iss m1v m2v m2t m2ts m4a m4b m4v mad mj2 mjpeg mjpg mka mkv mlp mm mmf mov mp+ mp1 mp2 mp3 mp4 mpc mpeg mpg mpga mpp mpu mve mvi mxf nc nsv nut nuv oga ogm ogv ogx oma ogg omg opus psp pva qcp qt r3d ra ram rl2 rm rmvb roq rpl rvc shn smk snd sol son spx str swf tgi tgq tgv thp ts tsp tta xa xvid uv uv2 vb vid vob voc vp6 vmd wav webm wma wmv wsaud wsvga wv wve
 [pcm]

Output plugins:
 shout null fifo pipe alsa ao oss pulse jack httpd recorder

Encoder plugins:
 null vorbis opus lame wave flac

Archive plugins:
 [bz2] bz2

Input plugins:
 file alsa archive curl ffmpeg smbclient nfs mms cdio_paranoia

Playlist plugins:
 extm3u m3u pls xspf asx rss soundcloud cue embcue

Protocols:
 file:// http:// https:// mms:// mmsh:// mmst:// mmsu:// gopher:// rtp:// rtsp:// rtmp:// rtmpt:// rtmps:// smb:// nfs:// cdda:// alsa://

```

Mpd starts playing dsf files as usual, but I get no sound.

```
$ lsusb -d 20b1

Bus 002 Device 002: ID 20b1:3023 XMOS Ltd

```

```
$ cat /proc/asound/DAC/pcm0p/sub0/hw_params

access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 176400 (176400/1)
period_size: 22050
buffer_size: 88200
```

```
$ cat /proc/asound/DAC/pcm0p/sub0/stream0

aune T1 USB DAC at usb-0000:00:12.2-1, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 3
    Packet Size = 228
    Momentary freq = 176376 Hz (0x16.0c08)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

```

Let me know if you need something else. Thanks!

lintweaker, 2016-02-20:
Everything looks good. The specs only refer to DSD using DoP.
Can you post the output of 'cat /proc/asound/DAC/stream0' ?

rhoeden, 2016-02-20:
Sure thing.
```
$ cat /proc/asound/DAC/stream0

aune T1 USB DAC at usb-0000:00:12.2-1, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 3
    Packet Size = 228
    Momentary freq = 176376 Hz (0x16.0c08)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
```

I tried adjusting my audio output in mpd.conf to dop, but still no sound.

```
$ cat mpd.conf

audio_output {
        type            "alsa"
        name            "Aune T1se [DSD]"
        device          "hw:2,0"
        dop             "yes"
        dsd_usb         "yes"
        # Enable native DSD playback
        dsd_native      "no"
        # Select 32-bit DSD_U32_BE output format
        dsd_native_type "2"
        auto_resample   "no"
        auto_format     "no"
        auto_channels   "no"
}

```

lintweaker, 2016-02-20:
stream0 output looks good. If DoP also does not work, that is really odd.
Better check with Aune, maybe a firmware issue? Can you check DoP using another OS (Win/Mac)?

rhoeden, 2016-02-20:
I tested it with foobar2000 on Win 8.1 and it just works. For some reason my mpd.conf works just fine with for example 24/192 flacs.

Also, while I was installing the drivers, I waded through the material that came with the T1se, and it would seem to be a dop device.

lintweaker, 2016-02-20:
Is there a DoP or DSD indicator on the DAC? If so we could test DoP on Linux with a simple DoP file using aplay.

rhoeden, 2016-02-21:
There are no indicator lights, but I got sound out of Deadbeef eventhough it’s choppy and cpu load rises to 50% (i7 @ 2.6GHz). With foobar2000 on Windows it was around 15% and without issues of choppiness or pops.

Does the recent mpd support dsd over pcm or should I stick with the dsd patched version while I’m trying to get this to work?

lintweaker, 2016-02-21:
Both the official MPD and my patched version support DoP. MPD itself should not be the issue as DoP has been working for years now and no changes have been made to that part.
There must something else amiss with your setup, no messages in the kernel log?

rhoeden, 2016-02-21:
This is getting mighty strange. I went through dmesg and journal, and some debugging with mpd, but there were no clues. Oddly enough the ncmpcpp’s visualizer works perfectly – at least I can see what the dsd sounds like.

I think I’ll try some live mpd distros to see if they work and try to reverse implement their configuration files to my setup.

Since it’s no longer relevant to the native dsd, I’ll close this issue. Thanks for pointing out the DoP thing since it relieves me from investigating why that doesn’t work either ;).

closed 2016-02-21 by rhoeden