lintweaker/xmos-native-dsd #10: Native DSD suport on Linux for WaveIO cards

WaveIO, 2016-02-24:
Hello Lintweaker,
I've followed your latest work on Linux, especially on Github and I tried to contact you on DIYaudio but got no response. I came here to ask for a bit of your time and help me add WaveIO board in the list of devices that are supported in Linux OS as per this link:
https://github.com/lintweaker/xmos-native-dsd

This WaveIO board does not have same USB VID as XMOS like many other competing boards do because it does provide a fully featured driver in Windows and for that, I was forced to acquire a Microsoft Authenticode certificate along with different USB VIDs than those used by XMOS.

In case are needed:
WaveIO USB VID = 0x20A0
WaveIO USB PID for HS USB Buses = 0x4143

I'm assuming that you only need PID for USB High Speed bus still I have the other one for FS if needed!

As for testing, I'm only using Windows... -as the worse thing happened in my life- but I have a friend of mine who can run whatever tests we have to conduct to fulfill this request.

Thank you for your time and patience,
Lucian (Lorien on DIYaudio forum)

lintweaker, 2016-02-24:
Hello,
I need some additional info before I can create a Linux kernel patch for further testing. Are you able to boot from a Linux live CD with the WaveIO connected and issue a few terminal commands?

I need output of:
cat /proc/asound/<your card>/stream0
lsusb

ljufa, 2016-02-29:
Hello, I'm a user of waveio on linux for a while so here is the command output on my machine. 

_uname -r_
4.2.0-27-generic #32-Ubuntu SMP Fri Jan 22 04:49:08 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

_cat /proc/asound/card1/stream0 _
AudioKit WaveIO USB Audio 2.0 at usb-0000:00:1d.0-1.1, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 63
    Momentary freq = 44101 Hz (0x5.8338)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

_lsusb _
Bus 002 Device 008: ID 058f:9540 Alcor Micro Corp. AU9540 Smartcard Reader
Bus 002 Device 005: ID 03f0:231d Hewlett-Packard Broadcom 2070 Bluetooth Combo
Bus 002 Device 004: ID 1bcf:2888 Sunplus Innovation Technology Inc. HP Universal Camera
Bus 002 Device 003: ID 1a81:1705 Holtek Semiconductor, Inc. 
Bus 002 Device 011: ID 20a0:4143 Clay Logic 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 007: ID 03f0:3a1d Hewlett-Packard 
Bus 001 Device 003: ID 138a:003c Validity Sensors, Inc. VFS471 Fingerprint Reader
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lintweaker, 2016-03-01:
Thanks, the output looks good. The wave IO is 20a0:4143 ?

WaveIO, 2016-03-01:
Hello Lintweaker,
 yes, as you can see on my first post!
 (Thank you for help ljufa!)
  Kind rregards,
  Lucian

norrest, 2016-03-11:
Edit quirks.c 
/* XMOS based USB DACs */
    switch (chip->usb_id) {
    case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
    case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
    case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */

add 

    case USB_ID(0x20a0, 0x4143): /* WaveIO USB Audio 2.0 */

recompile kernel or wait for patch! 
WaveIO - a you have clics/pop during play dsd?

ljufa, 2016-03-12:
Norrest, I manage to patch kernel and mpd-0.19.13 and hw_params file looking good, but no luck with getting sound. Only continuous hisss noise. Same results with HQPlayer...

Lucian, I guess this is waveio or dac related so we can continue discussion on diyaudio.

Regards.

cat /proc/asound/W20/pcm0p/sub0/hw_params 
access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100

norrest, 2016-03-12:
May be need add to altsetting == 3 ?

lintweaker, 2016-03-12:
Yes, alt setting 3 should be the correct one.

ljufa, 2016-03-12:
Moved case statement to the next block (altsetting==3) but getting the same result :(

Is there anything I need to change in alsa config? I'm running v1.0.29.

lintweaker, 2016-03-13:
You are running the latest firmware on the WaveIO?
What happens when you play DSD with DoP?

ljufa, 2016-03-13:
Yes I'm running the latest WaveIO_DSD256_Test2.bin firmware.

With mpd version 19.3 (patched and vanilla) when dop "yes" I'm getting this error in mpd log:

alsa_output: Failed to open "W20" [alsa]: Failed to configure DSD-over-PCM on ALSA device "hw:1,0"

also tried with 0.18 and was able to get sound...

norrest, 2016-03-13:
wrong settings!!!

dop "no"
dsd_native "yes"
dsd_native_type    "0"  or dsd_native_type    "2"

ALSO DoP dont work with DSD256

norrest, 2016-03-13:
lsusb show ok? 
Maybe this firmware only for dsd256)
Show  
cat /proc/asound 
cat /proc/asound/card..../pcm0p/

ljufa, 2016-03-13:
cat /proc/asound/W20/stream0 
AudioKit WaveIO USB Audio 2.0 at usb-0000:00:1d.0-1.2, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

- For config native dsd:

  dsd_native      "yes"
  dsd_native_type "2"
 # dop             "yes"

this is hw_params out 
cat /proc/asound/W20/pcm0p/sub0/hw_params 
access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 352800 (352800/1)
period_size: 32768
buffer_size: 131072

- For DoP config:
  # dsd_native      "yes"
  # dsd_native_type "2"
  dop             "yes"

mpd 0.19.0 is not able to open card so hw_params says closed. For 0.18 I think I was getting Format: S32_LE in this case.

Also tried all sample rate files:64,128 and 256 for all tests.

ljufa, 2016-03-18:
This is probably issue with my dac, it probably does not get switched to dsd mode correctly.
I hope I'll be able to play dsd after fix that.

norrest, 2016-03-19:
Where i can find firmware for WaveIO  ?

WaveIO, 2016-03-22:
@ norrest:
 here's the link for latest firmware release: 
http://www.luckit.biz/downloads/WaveIO_DSD256_Test2.bin
Kind regards,
 Lucian

WaveIO, 2016-04-08:
Any news with your DAC ljufa?

xxxbugxxxx, 2016-04-13:
maybe -Alsa version problem. I removed all the Alsa libs and solved issue for another different setup

norrest, 2016-04-13:
please run

grep VERSION_STR /usr/include/alsa/version.h

apt-cache show libasound2

ljufa, 2016-04-13:
Unfortunately no news from me, I'm still missing SPI controller that can switch my dac to dsd mode when needed :(

xxxbugxxxx, 2016-04-14:
this is mind

root@volumio:~# grep VERSION_STR /usr/include/alsa/version.h
#define SND_LIB_VERSION_STR     "1.1.1"
root@volumio:~# apt-cache show libasound2
Package: libasound2
Source: alsa-lib
Version: 1.0.28-1
Architecture: armhf
Maintainer: Debian ALSA Maintainers <pkg-alsa-devel@lists.alioth.debian.org>
Installed-Size: 938
Pre-Depends: multiarch-support
Depends: libc6 (>= 2.16), libasound2-data (>= 1.0.28-1)
Suggests: libasound2-plugins (>= 1.0.24)
Breaks: bluez-alsa (<= 4.94-2), libasound2-plugin-equal (<= 0.6-1), libasound2-plugins (<< 1.0.24)
Multi-Arch: same
Homepage: http://www.alsa-project.org/
Priority: optional
Section: libs
Filename: pool/main/a/alsa-lib/libasound2_1.0.28-1_armhf.deb
Size: 315956
SHA256: 76add8e7c134c2f012a4961a929dc1dd4edbac3ed2759d74a22803ac732ccaeb
SHA1: 10d45297f7584dd715f5f12afdabcd40a9b71a01
MD5sum: 10faafdeba6f2f4687dd38907490963c
Description: shared library for ALSA applications
 This package contains the ALSA library and its standard plugins, as well
 as the required configuration files.
 .
 ALSA is the Advanced Linux Sound Architecture.
Description-md5: f80c24ce80b76f9116fc6c7d1a8aa34b

xxxbugxxxx, 2016-04-14:
At least the stock Alsa version is low, won't work

norrest, 2016-04-14:
apt-cache show libasound2
Package: libasound2
Source: alsa-lib
Version: 1.0.28-1

- old version! 

Need 1.0.29 or latter

ljufa, 2016-06-08:
I made some progress with my dac today and was able to get dsd out of waveio on my speakers.
It was all fine between linux and waveio from the first attempt but dac was stuck in pcm mode.
Please let me know if you need additional info about current setup.

Regards.

WaveIO, 2016-06-11:
It's confirmed form here too! Lintweaker, if I can be of further help please let me know! Thank you!

lintweaker, 2016-06-12:
Okay, great. I'll create the patch.

lintweaker, 2016-06-12:
Patch is up!

ljufa, 2016-06-13:
I forgot to mention for me it is working with "altsetting == 2" case.

Lietweaker, from your previous experience when we can expect patch to be integrated into the stable kernel version?

lintweaker, 2016-06-13:
@ljufa did the firmware change recently? Can you post the output of:
[code]
cat /proc/asound/<card>/stream0
[/code]

ljufa, 2016-06-13:
Sorry, my mistake, it is altsetting ==3, I've built kernel couple of times so I've lost track of versions :)

-----------------------------------------

cat /proc/asound/card1/stream0 
AudioKit WaveIO USB Audio 2.0 at usb-0000:00:1d.0-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 56
    Momentary freq = 44102 Hz (0x5.8340)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

closed 2016-08-14 by lintweaker