lintweaker/xmos-native-dsd #11: Clicks/pops for XMOS DAC

norrest, 2016-02-29:
I can not in any way get the lack of clicks on my DAC, even using DSD_Native mode.
my Ustars Audio XMOS AK4495 dsd DAC work with your patch 
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100

I read that in contrast to DoP mode, DSD Native does not have these problems.
I ask not to close this task!

I wrote to the developers MPD (and email to ALSA devs)
http://bugs.musicpd.org/view.php?id=4320#c9858
But no one even wants to take on this problem!
I am sure that this problem is also related to this issue by implementing support DSD_Native!
As one of the variants of the solution, I see
add delay to 500ms in 

a) When start dsd files!
b) When rewinding dsd files! 
c) When switching dsd -> dsd or pcm and stop! 
Or add this variable in the configuration file!
 
I do not know who to contact! I have written to all of whom could have help , but nobody wants to deal with this problem. Thank you!

afalout, 2016-02-29:
Suggest you first eliminate your PC side from the problem:

1) Perform a fresh OS install, preferably a different OS from the one you have now. Ideally, Lintweaker will advise what OS and versions he is using so you know there is no issue there
2) Apply DSD patches/recompile, and DOCUMENT EVERY STEP SO IT CAN BE PEER REVIEWED
3) If the clicks persist, post a full document from step 2 here.

Hope this helps a bit
Andrej

norrest, 2016-02-29:
Please do not say that, what do not understand!

storrgie, 2016-02-29:
@norrest are you playing files converted by [sacd_extract](https://github.com/sacd-ripper/sacd-ripper/tree/master/tools/sacd_extract)? Because if so, there was a bug a while back that introduced a click/pop at the beginning/end of an extracted track. This was only for DSF extracted files if I remember right. I apologize for not being able to find the exact issue or patch, I just remember being plagued by it myself for a while.

On another note, you might consider being a less combative when someone like @afalout suggests that you provide a clear, review-able, set of steps for how your system is configured.

norrest, 2016-02-29:
DFF only!
I use kernels 4.0.- 4.4 
I sue mpd from 0.19.9 - 0.19.12
alsa tools and other 1.0.29 - alsa-lib-1.1.0

I've been fighting with this month! This is not a problem "Curves hands"

storrgie, 2016-02-29:
Does the pop occur at the beginning or end of a track (or basically when you change tracks) only with DFF files? If so, you're files were extracted from ISO to DFF with a version of sacd_extract that manifested that noise in the track. Further, if so, this has nothing to do with @lintweaker and I would significantly encourage him to close this issue.

norrest, 2016-02-29:
It does not matter, because when rewinding clicks as well! The problem is not in the files! Among them are those downloaded from official sources and are purchased from reliable suppliers!

norrest, 2016-03-01:
Add some info:

http://www.head-fi.org/t/800080/dsd-playback-pop-click#post_12383798

> The problem of muting the initial start is due to certain OS and apps - when the sample rate changes, the data in the buffers get sent out at the wrong rate. When you switch from DSD to PCM you then get a loud bang, or track played at the wrong rate for a period. Now this fault is in the app, its absolutely nothing to do with the DAC.
 
So to fix the apps problem, if the sample rate changes, I force a mute for 0.5 seconds, so that you do not hear these noises. On some apps, you can force a small silent period at the beginning and this solves the issue.
 
If I didn't do this I would have people complaining about bangs thumps and music being played at the wrong rate. Better silence than shots being fired off!  

>

lintweaker, 2016-03-01:
This issue seems very DAC dependent, I have a few DACs which do not exhibit this issue.
If this can be fix in MPD with one of the methods mentioned, it has to be added to the mainline MPD.

norrest, 2016-03-04:
As I understand it, refining MPD no one will do? I just can not understand why this no one does not want to deal! Highly tricky code? Or why? Maybe look for programmers? Can you clarify the situation in greater detail to whom it is best to apply? THANK U!!

lintweaker, 2016-03-04:
I merely maintain a patch which add more native DSD sample formats. I am not the author/maintainer of MPD. If you like this feature added to MPD please apply on the MPD development maillinglist.

closed 2016-03-04 by lintweaker

norrest, 2016-03-04:
Ok! if any want to help, please letter at the mpd-devel <@> musicpd.org