lintweaker/xmos-native-dsd #12: Amanero card now supporting DSD native

hifi25nl, 2016-03-03:
It seems that with new firmware Amanero Card is supporting DSD native in linux. These are the data given by Amanero:

OEM Combo384 default VID=0x16d0 PID=0x71A 

To enable/disable native DSD on Combo384 an USB Control request must be sent to endpoint 0. This procedure is supported on all firmware versions.
On firmware_1099rc1 an USB alternate settings "2" with a native DSD endpoint is supported.

The control request is of type Vendor to the Device.

bmRequestType=0x40, 
bRequest=0xB1, 
wLength=sizeof( struct VAD_DATA ), 
Data is a variable of type struct VAD_DATA

struct VAD_DATA {

	unsigned int command;
	unsigned int length;
	unsigned int buffer[64];
};

To enable native DSD 
command=0x8000  
length=0
buffer=undefined values
		
To disable native DSD
command=0x10000 
length=0	
buffer=undefined values

lintweaker, 2016-03-03:
I have looked into this before. It seems the newer firmware version is not available through their website (yet).
The new firmware supporting an additional alt setting sounds promising.

bibo01, 2016-03-03:
Pardon me, I am just interested in this issue and try to understand. Is there a problem with Amanero's code? It applies to all firmwares. Would you like Amanero to make firmware 1099rc1 public? How relevant is that to you? If that is an issue, I can ask Amanero directly to make it public.

lintweaker, 2016-03-04:
The method used in firmware 1099rc1 is preferable as it means only a very small patch to add support for it in Linux. So yes, public availability of the (final) version of that firmware would be great.

bibo01, 2016-03-04:
I spoke to "Amanero" and that firmware is already public by using the tool to update the firmware. However, he will make it visible on the web site too today.

lintweaker, 2016-03-04:
Thanks, 1099rc1 is now available through oemtool117. I flashed my Amanero Combo384 with it. Alt setting 2 is available:
`Amanero Technologies Combo384 Amanero at usb-0000:00:14.0-14.1, high speed : USB
Audio

Playback:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 2
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us`

I can now create a patch for it. Can you test? I have nothing connected to my Combo384 atm.

bibo01, 2016-03-04:
Sure - either me or hifi25nl can test it

lintweaker, 2016-03-04:
Patch is up (against current alsa-sound git). Uses default USB VID/PID. Firmware 1099rc1 required.

hifi25nl, 2016-03-04:
When I have time I will test (maybe tomorrow)

norrest, 2016-03-05:
Are need add at quirks.c and rebuilt kernel?

hifi25nl, 2016-03-05:
I have made a new realtime kernel with this patch for Archlinux (and audiolinux) here:
http://www.tophifi.it/ftp/packages/Amanero/

norrest, 2016-03-05:
command lsusb show 0x16d0, 0x071a ?

hifi25nl, 2016-03-05:
My audio status script is showing this:
card1	USB Audio card		usb3  --> 16d0:071a	Amanero Technologies Combo384

norrest, 2016-03-05:
its ok!

hifi25nl, 2016-03-05:
I have made a first test. I am sure that now I have the new firmware, since HQplayer is showing SMD option when not set to DoP

1) with old not patched kernel --> distorted sound (I can hear the song playing) on only one channel. the other is muted.
2) with new patche kernel --> distorted sound with both channels
3) with DoP option all is OK

hifi25nl, 2016-03-05:
Alsa i giving this playing a DSD64 file without upsampling

access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 22050
buffer_size: 44100

hifi25nl, 2016-03-05:
Some more information from /proc/asound/card1

Amanero Technologies Combo384 Amanero at usb-0000:00:1d.0-1.4.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 2
    Altset = 2
    Packet Size = 118
    Momentary freq = 88195 Hz (0xb.0640)
    Feedback Format = 16.16
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 2
    Altset 2
    Format: S32_LE DSD_U32_BE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

lintweaker, 2016-03-05:
Thanks for the report. The issue could with the sample format (not clear which format is supported by the Amanero) or a firmware issue.
When you play native DSD does the distorted sound resemble the music you are playing or complete noise?
You could try to swap out the sample format from DSD_U32_BE to DSD_U32_LE and change the player settings accordingly.

norrest, 2016-03-05:
do you try dsd mode =0 and 2 ?
what version alsa-lib/tools

hifi25nl, 2016-03-05:
> When you play native DSD does the distorted sound resemble the music you are playing or complete noise?
You could try to swap out the sample format from DSD_U32_BE to DSD_U32_LE and change the player settings accordingly.

Yes, there is sound but distorted (some high frequency added maybe)

In mpd-dsd I have changed to  dsd_native_type "3"
but how "swap out the sample format from DSD_U32_BE to DSD_U32_LE"?

norrest, 2016-03-05:
May be
```
  /* XMOS based USB DACs */
    switch (chip->usb_id) {
    case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
    case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
    case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */
        if (fp->altsetting == 2)
            return SNDRV_PCM_FMTBIT_DSD_U32_BE;
        break;
    case USB_ID(0x16d0, 0x071a): /* Amanero Combo384 */
    case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
    case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
    case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
    case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
    case USB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */
            if (fp->altsetting == 3)
            return SNDRV_PCM_FMTBIT_DSD_U32_BE;
        break;
    default:
        break;
    }
```

hifi25nl, 2016-03-05:
I have tried dsd_native_type "2", "0" and "3" in mpd-dsd. Only setting 2 give me sound on both channels. 

About altsetting, I think it must be 2, as said by manufacturer "On firmware_1099rc1 an USB alternate settings "2" with a native DSD endpoint is supported."

Maybe this is right?

     /* XMOS based USB DACs */
    switch (chip->usb_id) {
    case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
    case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
    case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */
        if (fp->altsetting == 2)
            return SNDRV_PCM_FMTBIT_DSD_U32_BE;
        break;
    case USB_ID(0x16d0, 0x071a): /* Amanero Combo384 */
        if (fp->altsetting == 2)
            return SNDRV_PCM_FMTBIT_DSD_U32_LE;
        break;
    case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
    case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
    case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
    case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
    case USB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */
            if (fp->altsetting == 3)
            return SNDRV_PCM_FMTBIT_DSD_U32_BE;
        break;
    default:
        break;
    }

bibo01, 2016-03-06:
I have used the patch compiled by hifi25nl as reported above.

Like him, **with new patched kernel I get distorted sound on both channels.**

Furthermore, I noticed that my DAC recognizes the incoming DSD signal always at a higher frequency. That is, if I am outputting DSD64 (with no resampling), the DAC sees it as DSD128; if I am sending DSD128, the DAC sees it as DSD256: and so on.

lintweaker, 2016-03-06:
Managed to get a test setup going, the sound is indeed recognizable but tinny. Switching between BE and LE sample formats does not seem to matter/improve things.
BTW word of warning: my version of MPD 0.19 lost its ability to use the 32-bit LE DSD sampleformat (compared to the MPD 0.18 version).

bibo01, 2016-03-06:
@ Lintweaker,

hifi25nl was telling me about a discrepancy between how DSD itself specifies samples in *bit*, while DOP and ALSA handle them as *bytes*. Hence, a factor of 8 or 16 has to be applied for the sample
rate configuration, according to the following table:

configured hardware
352.8kHz 705.6KHz 1411.2KHz <---- sample rate

8-bit 2.8MHz 5.6MHz 11.2MHz
16-bit 5.6MHz 11.2MHz

`-----------------------------------'
actual DSD sample rates

This would match my results of playback at a higher family - DSD128 instead of DSD64 and so on. Did you get similar results?
Above all, is it possible to adjust for it in the patch itself?

lintweaker, 2016-03-06:
@bibo01 the player side has to set the correct sample rate. Nothing new here.
For 32-bit DSD samples (contains 4x 8-bit DSD sample) the rate should be 88.2k for DSD64. That is half the rate using DoP (176.4k). For 8-bit DSD samples its 352.8k.

So far 32-bit DSD samples @ 88.2k (DSD64) works for XMOS based and other DACs.

We probably should ask Amanero what sample size, byte order etc they use. Maybe the use a different order for left right samples.

hifi25nl, 2016-03-06:
Amanero is modifying the firmware just now.

lintweaker, 2016-03-06:
Okay, thanks. I was about to sent them an e-mail. I'll wait for the updated firmware.

bibo01, 2016-03-07:
Great! It should be ready by tomorrow.

norrest, 2016-03-11:
bibo01 wake up!!!!!!!!

bibo01, 2016-03-11:
I spoke to Amanero a couple of days ago and I was expecting his firmware anytime, but it seems that it is not the case yet. Hopefully soon...

amanero, 2016-03-15:
hello i think the problem is that the data are in big endian. Is it possible to send in little endian?
Thank you!

amanero, 2016-03-15:
the data are in little endian, about the speed for DSD64 the firmware is expecting 44.1Khz but it receives a sample rate setup at 88.2Khz
is it possible to change it on the patch?

amanero, 2016-03-15:
the firmware works with data in little endian, for DSD64 is expected a sample rate setup at 44.1Khz but it receives 88.2Khz. Is it possible to change the sample rate on the patch?
Thank you!
Dom

lintweaker, 2016-03-16:
Hello @amanero ,
What sample size is the Amanero Combo384 expecting?
I am now using a 32-bit sample format
I have tried the LE sample format but there is no improvement over BE. The sound is still tinny with added noise (but recognizable) 
Also when using 44k1 instead of 88k2 the song is playing too slow.

bibo01, 2016-03-23:
@amanero ,

are we closer to a solution?

amanero, 2016-03-24:
Hello bibo01, there is firmware_1099rc2 available, data are in LE and it works at 88k2
Thank you!
Dom

norrest, 2016-03-24:
Do you have sound for dsd256?

amanero, 2016-03-24:
yes dsd256

norrest, 2016-03-24:
And do you have click/pops on forward, switch tracks and ect...?

norrest, 2016-03-24:
Like this http://bugs.musicpd.org/view.php?id=4320

amanero, 2016-03-24:
without a mute yes i have a switch, i can try to avoid

norrest, 2016-03-24:
`mute yes ?`
 where this parameter?

amanero, 2016-03-24:
i mean an hardware mute for PCM / DSD switch.

norrest, 2016-03-24:
Ok! Can you try in auto mode?

amanero, 2016-03-24:
or in auto mode yes

norrest, 2016-03-24:
Ok! I will wait!

hifi25nl, 2016-03-24:
I have made a test with MPD. All is OK!
HQPlayer however is crashing (Direct DSD or upsampling to DSD256).
Maybe is because HQPlayer is expecting BE format?

norrest, 2016-03-24:
As i understand you do not have clic`s/pops when foward/switch tracks? 
cat: /proc/asound/card0/
cat /proc/asound/Audio/pcm0p/sub0/hw_params 
please show!

hifi25nl, 2016-03-24:
access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100

Amanero Technologies Combo384 Amanero at usb-0000:00:1a.0-1.1.1, high speed : USB Audio

Playback:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 2
    Altset 2
    Format: S32_LE DSD_U32_BE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 

MPD-DSD
audio_output {
  type "alsa"
  name "Amanero"
  device "hw:1,0"
  dsd_usb "no"
  # Enable native DSD playback
  dsd_native "yes"
  # Select 32-bit DSD_U32_LE output format
  dsd_native_type "2"
  auto_resample 	"no"
  mixer_type      "disabled"
}

Yes I have clicks. I think that I must change the patch...

norrest, 2016-03-24:
In DoP mode click`s also? ( 
I'm starting be disappointed in mpd (
I try Windows in native mode, no clicks.
I do not know what to do. Bug Tracker mpd also did not respond.
He wrote to all who could, and ALSA developers too.
No one can help.

hifi25nl, 2016-03-24:
Yes also in DoP

norrest, 2016-03-24:
What is the meaning of this mode if it click`s?

amanero, 2016-03-24:
hello norrest, i remarked lot of zeros before a start, try to send 0xaaaaaaaa as silence pattern for DSD

norrest, 2016-03-24:
Hi! And it help? Add where?May be u can create patch for mpd? Or i some do not understand)

bibo01, 2016-03-25:
Where do I find firmware_1099rc2?

hifi25nl, 2016-03-25:
You can fin it in oem tool 117
But the patch on this site must be changed or not? Now it is sending BE format for Amanero.

lintweaker, 2016-03-30:
I have updated the patch to use the LE sampleformat. I did a few quick tests, the speed of the sound is now correct but I still get 'thinny' sound. I have tried a bunch of sample orders to no effect.
It would be nice to hear from @amanero what sample format/data order is expected.

hifi25nl, 2016-03-30:
I have made a kernel (realtime)  for Archlinux already patched here:
http://www.tophifi.it/ftp/packages/Amanero/
My results are strange. I tested with mpd-git and HQPlayer. HQPlayer is crashing. MPD is apparently working. Less noise but playing a DSD64 file the speed is completely wrong.
It is very accelerated. Yes, I have checked that quirks.c was correctly patched.
This is the alsa output:
access: RW_INTERLEAVED
format: DSD_U32_LE
subformat: STD
channels: 2
rate: 352800 (352800/1)
period_size: 32768
buffer_size: 131072

hicaoc, 2016-03-31:
I tested mpd-0.20 Play DSD64 also this phenomenon , rate: 352800 (352800/1), It is very accelerated. It should be 88200 . I use mpd-0.19.14 + native_dsd patch is normal.

lintweaker, 2016-03-31:
As mentioned in this thread before. The current version of MPD 0.19 + native DSD patch does not support the LE sample format. You can use my version of MPD 0.18 to test.

@hicaoc regular mpd 0.19/0.20 does not support native DSD using 32-bit sampleformats.

[503 more replies]