lintweaker/xmos-native-dsd #13: Alsa is giving format S32_LE with DSD stream

hifi25nl, 2016-03-16:
I  have just checked kernel 4.4.3-rt9-1-rt, mpd-git, Alsa 1.1.0 and iFi micro DSD DAC, playing a DSD 128 dsf file.
 
Both MPD and HQPlayer are working fine, led on iFi DAC is signalling DSD. Sound is right.
 
Alsa, strangely, is giving
 
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 352800 (352800/1)
period_size: 32768
buffer_size: 131072

Mpd configuration is 

audio_output {
	type		"alsa"
	name		"DSD"
	device		"hw:0,0"	# optional
	auto_resample 	"no"
	auto_format     "no"
	mixer_type      "disabled"
	dsd_usb		"yes"
}
 
But I am sure that only DSD is arriving to the DAC, since mpd cannot convert to PCM!
I have the same result using HQPlayer with the options DirectDSD, DoP, SDM

How do you explain this?

hifi25nl, 2016-03-16:
I have checked with standard archlinux kernel 4.4.5. Same results.
I would add also that in both cases I have only left channel (indistorted).

hicaoc, 2016-03-16:
dsd usb "yes" 
DoP mode is not converted to PCM, DSD data is encapsulated in PCM .
After you install the "DSD Native" patch for mpd ， the dsd native  signal can be directly output DSD

hifi25nl, 2016-03-16:
O, I will open a different issue and I will close this

closed 2016-03-16 by hifi25nl