lintweaker/xmos-native-dsd #14: DSD play broken on kernel 4.4.x?

hifi25nl, 2016-03-16:
Kernel realtime 4.4.3_rt9
Kernel standard 4.4.5
Alsa 1.1.0
Distribution: Archlinux

Is seems that DSD play is broken on these kernels (at least DoP, maybe also native)

In two different computers checked with MPD and HQPlayer
iFi micro iDSD led is receiving DSD signal, but right channel is muted.

norrest, 2016-03-18:
No! I Use kernel 4.4.3 - 4.4.4  - 4.4.5 
And in both channels was sound!

norrest, 2016-03-18:
http://ifi-audio.com/micro-idsd-ifi-xmos-firmware/
18 Feb. 2016	Major optimisations
• All-new XMOS ‘deep-core processes’ software for different behaviour of SPDIF (after XMOS previously modified the USB section).
• Enables DSD256 for 48kHz based sample rates up-conversion when using DSD256 in DoP mode.
• Fixed issues surrounding DoP standard; including ‘clicks’ when changing tracks and sonic differences to native DSD and other minor bug fixes.
• Minor updates/fixes

hifi25nl, 2016-03-20:
I have updated the firmware on the micro iDSD. Same problem. I tried also kernel 4.1.x.
I discovered that I have only one channel also on PCM. Another PCM DAC is working fine.
So I think this is not a specific problem with DSD play. Maybemy my iDSD is broken on the headphone output...

closed 2016-03-20 by hifi25nl

norrest, 2017-05-27:
So how it now? DSD)native play fine?