lintweaker/xmos-native-dsd #16: Integration in Volumio

volumio, 2016-04-02:
I really love your work! What do you think if we work together to integrate this in Volumio?

lintweaker, 2016-04-02:
Thanks! What do you have in mind?

volumio, 2016-04-02:
I was thinking on a build script to compile kernel with this patches integrated and alsa lib , while I'll take care of mpd. I was mainly thinking to pi and x86...
What do you think?

hicaoc, 2016-04-05:
Good job,

I am now a manual patch to Volumio , and can be integrated together , then that's great！

volumio, 2016-04-05:
Interesting, if you could share your manual patch, that could help as a starting point...

hicaoc, 2016-04-05:
@volumio ，Each DAC may have separate patches, I patched my own DAC.
If you put all the patches are written script, you may need to work a lot.

hicaoc, 2016-04-05:
After installing volumio 2 rc1 , I 'll upgrade alsa-lib and mpd support native-dsd

#!/bin/sh

apt-get install make gcc patch

upgrade alsa-xxx  1.0.28 to to 1.1.0 

apt-get purge libasound2 libasound2-data libasound2-dev  alsa-base alsa-utils 
wget  ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.1.tar.bz2 
wget  ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.1.tar.bz2
wget  ftp://ftp.alsa-project.org/pub/tools/alsa-tools-1.1.0.tar.bz2
tar -jxf alsa-lib-1.1.1.tar.bz2 
tar -jxf alsa-utils-1.1.1.tar.bz2
tar -jxf alsa-tools-1.1.0.tar.bz2
cd alsa-lib-1.1.1 ; ./configure ; make -j4 ; make install 
cd ../alsa-utils-1.1.1 ; ./configure ; make -j4 ; make install  
cd ../alsa-tools-1.1.0 ; ./configure ; make -j4 ; make install
cd .. 

wget http://www.musicpd.org/download/mpd/0.19/mpd-0.19.14.tar.xz
apt-get purge mpd
tar -xf mpd-0.19.14.tar.xz
git clone  https://github.com/lintweaker/mpd-dsd-019.git
cp mpd-dsd-019/0001-Add-native-DSD-support-to-MPD-0.19.14.patch mpd-0.19.14
cd mpd-0.19.14 ; patch -p1 < 0001-Add-native-DSD-support-to-MPD-0.19.14.patch 
./autogen.sh ;./configure ; make -j4 ; make install 
cd ..

/etc/init.d/mpd restart

volumio, 2016-04-05:
Great, I saw that for >3.19 the Kernel patches are mainline... That makes it easier to integrate! 
And with your instructions it will be a breeze... I'll let you know! 
thanks

xxxbugxxxx, 2016-04-20:
@volumio 

Native dsd kernel patch for xmos interface incorporated into hard kernel 's odroid c2's kernel repo. 
:)

volumio, 2016-04-20:
It's a request or a notification?
I'm doing the same now for pi and x86...

volumio, 2016-04-20:
@hicaoc Do you have a simple method to create .deb packages for the compiled utils?

xxxbugxxxx, 2016-04-20:
@volumio

I have got the pull request in for the c2 hard kernel Linux repo coz it is still on 3.14.

Since it is already mainlined since 3.19 so should be fine for other platforms.

hicaoc, 2016-04-21:
There is no make .deb, the compilation process there are many related problems to solve. It takes a long time

norrest, 2016-04-21:
**volumio** 
Hi! 
What you need to integrate? You just need a new kernel, new ALSA and patched mpd!

I've already done it for version 1.5 *
But version 2.0 - is crude and does not have the support of many boards

volumio, 2016-04-21:
I'm trying now to use sid's alsa packages... From your guide they should be ok... 
Is that right?
https://packages.debian.org/it/source/sid/alsa-lib

So I need to compile only mpd 
(no problems about compiling, is more a issue related with Volumio's build script)

closed 2017-02-24 by lintweaker