lintweaker/xmos-native-dsd #17: PS Audio DirectStream DAC

norrest, 2016-04-21:
Hello! There is an idea to check the Native mode with this DAC. But unfortunately it locally, I can not do that. I can only prepare  OS system in advance and send the person to be tested.
Maybe someone will tell USB_ID?
Maybe some like SB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */

sonorejr, 2016-04-21:
I checked the DirectStream and the DirectStream JR and they are can't work as configured by PS Audio.

norrest, 2016-04-21:
can u tell me USB_ID?
lsubs command

sonorejr, 2016-04-21:
2616:0110 It doesn't work. Also that DAC will convert native DSD back to DoP as part of it's internal process. Your better off just leaving in DoP format so the DAC doesn't have to do the extra work.

norrest, 2016-04-21:
> Convert native DSD back to DoP as part of it's internal process
Do you have proofs? 
Thank you!!!!!!

sonorejr, 2016-04-22:
I asked Ted S. about it via e-mail.

norrest, 2016-04-22:
Ok, if he answer, please write!

norrest, 2016-04-28:
Hi @sonorejr !
Do you have any news?

closed 2016-06-23 by norrest