lintweaker/xmos-native-dsd #22: Patch for DSD only dacs

hifi25nl, 2016-10-01:
I am trying to patch quirks.c for DSD native output ONLY. Since this DAC is using the Gustard USB interface I have made this modification:

case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
return SNDRV_PCM_FMTBIT_DSD_U32_BE;
break;
case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
case USB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */
if (fp->altsetting == 3)
return SNDRV_PCM_FMTBIT_DSD_U32_BE;
break;
default:
break;
}

Result: DoP is working, but native DSD output only noise

Some suggestions?

lintweaker, 2016-10-01:
For DoP you do not need to patch, it uses the PCM sample formats.
Regarding the noise when you play DSD, it could one of the following:
* Wrong DSD sample format, your DAC needs SNDRV_PCM_FMTBIT_DSD_U32_LE instead of SNDRV_PCM_FMTBIT_DSD_U32_BE
* Wrong alt setting
* Your DAC needs additional USB cmds to switch between PCM and DSD

Miveraaudio, 2016-10-02:
Thanks for the info. If we get the settings right, will media player applications such as Hqplayer recognize the DAC as being only native DSD capable?

hifi25nl, 2016-10-03:
The problem here is this:
The USB interface supports both PCM and DSD, but the DAC is DSD only. 
What trick can I use so that a software like HQPlayer will recognize the DAC as DSD only and so will not default to PCM? (this is important for a headless system playing in DSD with Roon)

My previous solution is not working, but I hope a patch could do this. The alternative maybe would be to change the USB interface firmware.

lintweaker, 2016-10-03:
With a custom kernel patch it should be possible to only populate the DSD sample format. Modifiying the firmware is probably a better option (if possible), it should only advertise the 'SPECIAL' sample format (not sure if this within the USB AUDIO 2 spec).

Miveraaudio, 2016-10-03:
Thanks. Modifying the firmware to accomplish this is not possible. It must be done with an ALSA patch. 

Here's what our USB interface engineer said regarding this:

So if you want to default back to DSD mode the Linux driver needs to be modified to show only DSD mode in the playback options. 

"USB AUDIO CLASS 2 has pcm which is default and cannot be changed. The PCM also sets the sample rate for DSD. So 352KHz pcm equals DoP128 and native DSD 256.

DoP uses pcm has container where an toggling byte indicates that DSD data is transmitted. For native DSD a data format called RAW is used."

Do you know what needs to be entered in to tell the media player that the DAC is native DSD only? I'm sure what's missing from the first mod was the altsetting info. It seemed to play both RAW and PCM simultaneously with that setting.

Miveraaudio, 2016-10-03:
I got an explanation of what's actually happening with our first mod from JLakko, but no solution on how to do it right :(

>  it'll send only DSD, but the DAC thinks it is receiving PCM while it is in fact receiving DSD. You need to make the driver always use the DSD altsetting towards DAC and not advertise any PCM formats towards applications.

hifi25nl, 2016-10-03:
How?.... It should have only Altset 3, a firmware change.

Piero

On 10/03/2016 09:39 PM, Mike wrote:
> You need to make the driver always use the DSD altsetting towards DAC 
> and not advertise any PCM formats towards applications.

-- 

Piero Olmeda su Facebook <https://www.facebook.com/pieroolmeda>
Blog di Piero Olmeda <http://pieroolmeda.blogspot.it>
Piero Olmeda su Google+ 
<https://plus.google.com/117349928118716099173/posts>
Piero Olmeda su Linkedin 
<https://www.linkedin.com/pub/piero-olmeda/a5/102/b66>
Profilo autore su Amazon <http://www.amazon.com/Piero-Olmeda/e/B00YEG8VO8>
Profilo autore su Goodreads 
<https://www.goodreads.com/author/show/14038972.Piero_Olmeda>

Logo <http://audio-linux.com>

Miveraaudio, 2016-10-03:
When you insert the "break"

> case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U /
> return SNDRV_PCM_FMTBIT_DSD_U32_BE;
> break;
> case USB_ID(0x20b1, 0x2009): / DIYINHK DSD DXD 384kHz USB to I2S/DSD /
> case USB_ID(0x20b1, 0x2023): / [...]

after "USB_ID(0x20b1, 0x000a" is that not separating the"if (fp->altsetting == 3)" from being applied to "USB_ID(0x20b1, 0x000a" 

Perhaps there's a different altsetting command than  "if (fp->altsetting == 3)" that needs inserted after "USB_ID(0x20b1, 0x000a" rather than nothing.

closed 2016-10-22 by hifi25nl