lintweaker/xmos-native-dsd #23: No DSD_U32_BE when using Aune S16

archphile, 2016-11-07:
Hi,

I have an Aune S16. I'ts XMOS receiver appears to be the same as the Aune X1S which has support on the linux kernel:

lsusb gives the following:

lsusb -d 20b1:
Bus 001 Device 006: ID 20b1:3023 XMOS Ltd

If I have understood correctly, There should be DSD_U32_BE for this receiver too, however aplay -l output is:

Card 1, ID `DAC', name `S16 USB DAC'
  Device 0, ID `USB Audio', name `USB Audio', 1 subdevices (1 available)
    2 channels, sampling rate 44100..384000 Hz
    Sample formats: S32_LE

BTW, I am also an owner of a DIYINHK XMOS based dac, where everything is fine:

Card 1, ID `D20', name `DIYINHK USB Audio 2.0'
  Device 0, ID `USB Audio', name `USB Audio', 1 subdevices (1 available)
    2 channels, sampling rate 44100..384000 Hz
    Sample formats: S16_LE, S32_LE, SPECIAL, DSD_U32_BE.

I use latest odroid kernel which as far as I know has latest code backported with regards to the DSD format:

https://github.com/hardkernel/linux/blob/odroidc2-3.14.y/sound/usb/quirks.c

Regards,

Michael

lintweaker, 2016-11-10:
The referenced quirks.c file seems fine. Can you show the output of 'cat /proc/asound/<card>/stream0' for your DAC?

archphile, 2016-11-11:
This is the output when listening to dsd64 with standard mpd 0.19.19:

cat /proc/asound/card1/stream0 
aune S16 USB DAC at usb-dwc2_b-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 228
    Momentary freq = 176451 Hz (0x16.0e70)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 2 IN (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

and this is the output with mpd using your native-dsd patch:

cat /proc/asound/card1/stream0 
aune S16 USB DAC at usb-dwc2_b-1.2, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 1
    Packet Size = 118
    Momentary freq = 88226 Hz (0xb.0738)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 2 IN (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

Btw in last case with your patch, what I hear is only whitenoise!

Regards,

Michael

lintweaker, 2016-11-11:
The patch for the Aune X1S applies to alt set 3, your DAC does not have alt set 3 so hence no native DSD support. You can change the patch for yourself to work on alt set 2

archphile, 2016-11-11:
Thank you very much for your reply!

Is there any chance for official support of the Aune S16 if I provide you with all necessary feedback?

Thank you in advance.

Many thanks,

Michael

lintweaker, 2016-11-11:
Unfortunately official support is not possible as both DACs share the same USB ID. If I create a patch for your DAC then X1S will also get DSD enabled on alt 2 which does not work on the XS1. Too bad Aune did not use seperate IDs.

archphile, 2016-11-11:
Completely understood.

One last question: Will this patch be everything I need for my case?

https://github.com/archphile/linux-1/commit/d50eb765430e345d92535631b19c4a54eb835c4d.patch

Many thanks,

Michael

archphile, 2016-11-11:
I applied the patch above with no luck. It now shows DSD_U32_BE support but still whitenoise. Anyways, there's no need for an open issue.

Thanks for everything!

archphile, 2017-04-24:
I am sorry for opening this issue again but I am hoping for some advice.

I did a fresh installation of ArchlinuxARM and recompiled udoo quad kernel with the following patch:

http://archphile.org/lab/s16/s16.patch

The kernel is 4.10

alsacap is now showing DSD format support:

```
*** Scanning for playback devices ***
Card 0, ID `DAC', name `S16 USB DAC'
  Device 0, ID `USB Audio', name `USB Audio', 1 subdevices (1 available)
    2 channels, sampling rate 44100..384000 Hz
    Sample formats: S32_LE, DSD_U32_BE
      Subdevice 0, name `subdevice #0'
```
cat /proc/asound/card*/pcm*p/sub*/hw_params

```
access: RW_INTERLEAVED
format: DSD_U32_BE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100

```

But I am still getting a mixture of whitenoise/weird sounds I can't describe. Is there anything else I could try?

Thank you in advance.

Regards,

Michael

lintweaker, 2017-04-25:
Are you sure your device supports native DSD? Looking at the specs I see only DoP mentioned.

archphile, 2017-04-25:
Thanks for your reply.

I have emailed Auneaudio and currently waiting for their reply.

I just wanted to know if technically I have done everything needed!

archphile, 2018-01-13:
I am really sorry for bumping a very old "issue", but as I was about to give up, I contacted Auneaudio and asked them if my dac supports native dsd under Windows, and the reply was positive.

During all these months, I did various kernel builds (always patching kernel), hoping that I will make native dsd work, but with no lack. Just a last question before I give up:

Is there anything else I should consider modifying and rebuild?

- this dac has the same usb id with Aune X1S but with altset 3, so I delete X1S line and put a new one under altset 3.

- should I try a different format than DSD_U32_BE? 

- Is there anything else I should consider trying?

The only reason I want to make native dsd functional is because its DoP implementation really sucks and it's full of pops/clicks etc when changing tracks, repositioning tracks etc and as I see, even alsa has implemented the correct dsd (0x69) silence for various native dsd formats:

https://patchwork.kernel.org/patch/9470981/

As I understand this nice patch does not affect dop, so we have another reason to need native dsd.

Thank you in advance,

Michael