lintweaker/xmos-native-dsd #31: DSD_U32_BE really gone in new alsa/kernel??

norrest, 2017-05-27:
I have new DAC
(0x27f7, 0x1301): /*  like  Wyred DSD
add to 
	case USB_ID(0x29a2, 0x0086): /* Mutec MC3+ USB */
	case USB_ID(0x20b1, 0x2005): /* Denafrips DAC */
	case USB_ID(0x27f7, 0x3002): /* W4S DAC-2v2SE */
	case USB_ID(0x27f7, 0x1301): /* Wyred DSD Interface */
		if (fp->altsetting == 3)
			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
		break;
	default:
		break;

But allways  S32_LE

norrest, 2017-05-27:
cat /proc/asound/card*/pcm*p/sub*/*
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100
card: 1
device: 0
subdevice: 0
stream: PLAYBACK
id: USB Audio
name: USB Audio
subname: subdevice #0
class: 0

uname -r
4.4.8

 apt-cache show alsa-tools
Package: alsa-tools
Status: install ok installed
Priority: extra
Section: sound
Installed-Size: 177
Maintainer: Debian ALSA Maintainers <pkg-alsa-devel@lists.alioth.debian.org>
Architecture: armhf
Version: 1.1.0-1

 apt-cache show libasound2
Package: libasound2
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 800
Maintainer: Debian ALSA Maintainers <pkg-alsa-devel@lists.alioth.debian.org>
Architecture: armhf
Multi-Arch: same
Source: alsa-lib
Version: 1.1.0-1
Depends: libc6 (>= 2.16), libasound2-data (>= 1.1.0-1)
Suggests: libasound2-plugins (>= 1.0.24)
Breaks: bluez-alsa (<= 4.94-2), libasound2-plugin-equal (<= 0.6-1), libasound2-plugins (<< 1.0.24)
Description-en: shared library for ALSA applications

norrest, 2017-05-27:
Earled i so
Altset 1
Altset 2
Altset 3

But now its only     Altset = 1

cat /proc/asound/card1/stream

Playback:
  Status: Running
    Interface = 2
    Altset = 1
    Packet Size = 112
    Momentary freq = 88192 Hz (0xb.0628)
    Feedback Format = 16.16
  Interface 2
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 5 OUT (ASYNC)
    Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us

norrest, 2017-05-28:
i

closed 2017-05-28 by norrest

ronnyzhao, 2018-05-16:
the same question, give me help,email: zybroseman2010@126.com