lintweaker/xmos-native-dsd #32: Moon Neo 230HAD support

oosten246, 2017-05-31:
Would it be possible to add support for the Moon Neo 230HAD?

cat /proc/asound/card1/stream0

Simaudio Ltd. MOON USB DSD HD Audio at usb-lm1-1.2, high speed : USB Audio
Playback:
   Status: Running
     Interface = 2
     Altset = 1
     Packet Size = 448
     Momentary freq = 352910 Hz (0x2c.1d20)
     Feedback Format = 16.16
   Interface 2
     Altset 1
     Format: S32_LE
     Channels: 2
     Endpoint: 5 OUT (ASYNC)
     Rates: 32000, 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
     Data packet interval: 125 us

lsusb -s 1:3 -v

Bus 001 Device 003: ID 2397:000d
Device Descriptor:
   idVendor           0x2397
   idProduct          0x000d
   iManufacturer           1 Simaudio Ltd.
   iProduct                2 MOON USB DSD HD Audio

lintweaker, 2017-06-02:
The output suggests it only supports PCM. There is only one altset, normally there are at least 2.

oosten246, 2017-06-02:
I use the DAC with Roon. On a Linux endpoint it plays DSD128 via DoP. On a Windows endpoint using the ASIO driver the DAC plays DSD256.

lintweaker, 2017-06-03:
It probably uses vendor specific USB commands to switch between PCM and DSD.

oosten246, 2017-06-03:
So, I am out of luck then?