lintweaker/xmos-native-dsd #33: Support of the Armature Hecate interface

patairmax, 2017-12-05:
The Armature Hecate interface is closed to (maybe even identical to) the Singxer F-1.
It support up to native DSD 256 on its USB input. It has an I2S output which always produce a native DSD data flow. I'm using it with success up to DSD 128 with a NanoPI Neo or a RPI3 with DietPI and RoonBrdige. The Roon support team told me to look at this project, and it seems that here is a solution.
However the Armature Hecate is not in the supported list. The command lusb produce the following:
root@DietPi:~# lsusb -d 20b1:
Bus 001 Device 002: ID 20b1:3086 XMOS Ltd 

I noticed that 20b1:3086 is neither 20b1:3008 nor 20b1:2009.

Not sure it is the correct place (by raising an issue here) to ask this update, correct me if I'm wrong, sorry in advance.

Is it possible to have an update of the lintweaker/xmos-native-dsd project that would support the Armature Hecate board ?

Thanks and congrats for this initiative 👍

patairmax, 2017-12-06:
Actually I would ask my question more precisely. 
I'm wondering if it is necessary to have an update of the lint weaker/xmos-native-dsd files in order to enable Native DSD on a Linux Kernel 4.11 with the Hecate Board ?

Here some other info:
root@DietPi:~# uname -a
Linux DietPi 4.11.12-sun8i #2 SMP Wed Nov 22 13:50:32 CET 2017 armv7l GNU/Linux

root@DietPi:~# cat /proc/asound/card1/stream0
Carnyx TREXO at usb-1c1d000.usb-1, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: S16_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us

root@DietPi:~# lsusb -d 20b1:
Bus 001 Device 002: ID 20b1:3086 XMOS Ltd

Thanks

lintweaker, 2017-12-08:
With your last info I should be able to create a patch for native DSD support for your DAC. I'll upload it today.

closed 2017-12-08 by lintweaker

patairmax, 2017-12-11:
Many thanks. I'll try to fix my kernel with the patch. I hope to not have difficulties with DietPI OS which is already in Kernel 4.11 in regards to the Kernel you are talking about 3.8.x if I'm not wrong...