lintweaker/xmos-native-dsd #34: Global patch create

norrest, 2018-01-13:
Hi! Can you crate global patch for all dsd dacs?

norrest, 2018-01-13:
**for v4.14 kernel**

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/sound/usb/quirks.c?h=v4.14.13

norrest, 2018-01-13:
Its Ok?
[patch.txt](https://github.com/lintweaker/xmos-native-dsd/files/1628290/patch.txt)

norrest, 2018-01-13:
[patch.txt](https://github.com/lintweaker/xmos-native-dsd/files/1628291/patch.txt)
Its ok?

lintweaker, 2018-01-13:
It is still on my todo (low prio). The items need to be in order, lower id's first. But a good first attempt :-)

norrest, 2018-01-13:
This patch dont work? (

norrest, 2018-01-13:
case USB_ID(0x20b1, 0x3086): /* Singxer F-1 converter board */
case USB_ID(0x20b1, 0x3086): /* ARMATURE Hecate */

norrest, 2018-01-13:
[dsd_native.txt](https://github.com/lintweaker/xmos-native-dsd/files/1628315/dsd_native.txt) So?

norrest, 2018-01-13:
So? 
[dsd_native.txt](https://github.com/lintweaker/xmos-native-dsd/files/1628316/dsd_native.txt)

lintweaker, 2018-01-13:
almost, you also need to order the second number (so first on manufacturer id, then on device id).

norrest, 2018-01-13:
there are numbers and letters, how to correctly ottortirovat? This is generally important? Or just for beauty! Can you help?
Is there a ready-made file?

closed 2018-01-13 by norrest