lintweaker/xmos-native-dsd #35: SMSL M8A support

hifi25nl, 2018-06-14:
Card specifications:

card1   USB Audio card   usb6  --> 152a:85dd     SMSL SMSL M8A

The output of this dac from /proc/asound is

``SMSL SMSL M8A 1.1.2 at usb-0000:00:1d.0-1.3, high speed : USB Audio
Playback:
 Status: Stop
 Interface 1
 Altset 1
 Format: S32_LE
 Channels: 2
 Endpoint: 1 OUT (ASYNC)
 Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
 Data packet interval: 125 us
 Interface 1
 Altset 2
 Format: S32_LE
 Channels: 2
 Endpoint: 1 OUT (ASYNC)
 Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
 Data packet interval: 125 us
 Interface 1
 Altset 3
 Format: SPECIAL
 Channels: 2
 Endpoint: 1 OUT (ASYNC)
 Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000 Data packet interval: 125 us

It should have support for DSD native up to DSD512.  The patch should use Altset3?

lintweaker, 2018-06-14:
That looks about right

hifi25nl, 2018-06-27:
I confirm that this is working with
`case USB_ID(0x152a, 0x85dd): /* SMSL M8A */
 		if (fp->altsetting == 3)
 			return SNDRV_PCM_FMTBIT_DSD_U32_BE`