lintweaker/xmos-native-dsd #41: Is this patch still necessary with Ubuntu 18 and ALSA 1.1.9?

Juno808, 2019-08-23:
Hi,

Is this patch still necessary with Ubuntu 18.04 and ALSA 1.1.9?

Do we still need to compile Kernel to add native DSD support?

Thank you.

Juno

lintweaker, 2019-08-23:
Hi,
Not sure which path you are referring to but ALSA has been DSD ready from quite some years and a lot of device specific DSD patches have made it into the kernel

Juno808, 2019-08-26:
Hi lintweaker,

Thanks for your reply. And thanks for your greats works related to DSD.

I was referring to patches for Kernel. I am using Kernel 5.0.0-25-Generic from Ubuntu 18. Do I need to recompile the Kernel to include ALSA drivers for DSD? Or would it just work?

Thank you.

Juno

lintweaker, 2019-08-29:
Hi,
I was referring to the kernel patches as well. A lot of them (if not all and more) are in the stock kernel.

closed 2019-08-29 by lintweaker