lintweaker/xmos-native-dsd #42: Native DSD support for FiiO K3

mluizescu, 2019-11-01:
Hi, could you add native DSD support for FiiO K3? It is based on XMOS U30881C10. Under Linux, it only supports DoP up to DSD128. Under Windows, it supports native DSD up to DSD256. The usb ID is 2972:0047.

Thank you.

dc740, 2019-11-19:
Hi,
I sent a patch already. You can compile your own kernel to add support, it's pretty straight forward.
Check the mailing list:
https://lkml.org/lkml/2019/11/19/886

Instead of adding the VID and PID, I used the VID together with the auto-detect generic method, so other FiiO devices will be supported too.

-or at least the ones with the same VID. Apparently they have different vendor IDs depending on the product.-

dc740, 2020-01-19:
@mluizescu  can you test the patch and comment on the mailing list? Looks like it has been largely ignored

mluizescu, 2020-01-19:
I tested the patch today on Archlinux and it worked. I am able to play DSD
files up to DSD256, the maximum supported by the sound card. I only
compiled the snd-usb-audio module with the patch and replaced the stock one.

mluizescu, 2020-01-19:
I tested the patch today on Archlinux and it worked. I am able to play DSD
files up to DSD256, the maximum supported by the sound card. I only
compiled the snd_usb_audio module with the patch and replaced the stock one.
DSD playback was confirmed by the green LED on the soundcard and by checking /proc/asound (altset 3 is used during DSD playback).

closed 2020-01-19 by mluizescu

dc740, 2020-01-19:
@mluizescu can you reply to the mailing list with a Tested-by tag? so the maintainers can see the change has been tested at least twice. I didn't get any replies, and the patch was not merged on any branch. I'm afraid it may get lost or ignored. 
https://www.kernel.org/doc/html/v5.4/process/submitting-patches.html#using-reported-by-tested-by-reviewed-by-suggested-by-and-fixes

thanks

mluizescu, 2020-01-19:
@dc740 can you tell me how I can reply? I'm not on LKML and have no experience with it.

schlunsen, 2020-02-17:
Did it get merged?

dc740, 2020-07-18:
It was not merged, and there is no activity either. So I don't know if there are any reasons at all. I'll probably reply to the patch in a couple of months if it doesn't get any reply

amadejkastelic, 2020-12-15:
Any news on this?

dc740, 2020-12-28:
It finally got merged!
They got the commit author wrong...but...it happens. I'm used to it.

At least this will be supported on newer kernels. It was also backported to
the latest 4 point release.