lintweaker/xmos-native-dsd #43: Support for Primare DM35 DAC?

christer74, 2020-01-26:
Can the Primare Prisma DAC be supported?

First rows of lsusb -v;
Bus 001 Device 002: ID 24a4:0002  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x24a4 
  idProduct          0x0002 
  bcdDevice            6.f2
  iManufacturer           1 Primare
  iProduct                3 Primare_DM35_v1.06
  iSerial                 0 
  bNumConfigurations      2

Playing a ordinary 16 bit FLAC-file;

cat /proc/asound/card1/stream0
Primare Primare_DM35_v1.06 at usb-0000:00:14.0-3, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 2
    Packet Size = 72
    Momentary freq = 44102 Hz (0x5.8340)
    Feedback Format = 16.16
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us
    Bits: 16
  Interface 1
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us
    Bits: 32
  Interface 1
    Altset 3
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000
    Data packet interval: 125 us
    Bits: 32

kimmot, 2020-02-04:
I could provide you a special test version of Daphile with an experimental native DSD support if you would be willing to test it. Please contact me using the form at www.daphile.com

christer74, 2020-02-05:
OK. Just used the contact form.